osrf/subt #1064: some problem in installation for ROS Noetic

lzhhh6161, 2024-12-08:
when i install  subt following the tutorial :https://github.com/osrf/subt/wiki/Installation-for-ROS-Noetic, i can not understand the step"Renamed fuel.gazebosim.org to fuel.ignitionrobotics.org in cache", i can not find any config file, can someone share the file?

peci1, 2024-12-08:
I think what was meant was:

    mv ~/.ignition/fuel/fuel.gazebosim.org ~/.ignition/fuel/fuel.ignitionrobotics.org

But maybe a better and more future-proof solution would be:

    rm -rf ~/.ignition/fuel/*  # BEWARE: This deletes all your downloaded models
    mkdir ~/.ignition/fuel/fuel.gazebosim.org
    ln -s ~/.ignition/fuel/fuel.gazebosim.org ~/.ignition/fuel/fuel.ignitionrobotics.org

Could you try either of these and report back?

lzhhh6161, 2024-12-09:
> I think what was meant was:
>
> ```
> mv ~/.ignition/fuel/fuel.gazebosim.org ~/.ignition/fuel/fuel.ignitionrobotics.org
> ```
>
> But maybe a better and more future-proof solution would be:
>
> ```
> rm -rf [...]

I found all the required models are located in fuel.gazebosim.org, so i tried:

```
rm -rf ~/.ignition/fuel/fuel.ignitionrobotics.org/openrobotics/models
ln -s ~/.ignition/fuel/fuel.gazebosim.org ~/.ignition/fuel/fuel.ignitionrobotics.org
```

When i start the sim, it still download required model to ~/.ignition/fuel/fuel.ignitionrobotics.org/openrobotics/models . Actually, I am now able to successfully run the simulation, but the simulation has a long loading time and real time factor is only about 13%.

closed 2024-12-09 by lzhhh6161

peci1, 2024-12-09:
Thanks. I forgot to add to this 2nd alternative that you have to run the `ign download` command again after you create the symlink.

Regarding the RTF, as already said, SubT sim was never designed to achieve or even get close to 100%.