osrf/subt #1065: Lidar cannot publish /cloud topic

jiaminguo1995, 2024-12-11:
I try to get the pointcloud. But the Lidar seems not work, the topic of the pointcloud did not  be published. I check the log and did not found the error related to this issue. 

Here are two logs I guess may be related to this issue, but I am not sure.
Error Code 13 Msg: Directory doesn't exist[/home/jamie/.ignition/fuel/fuel.gazebosim.org/openrobotics/models/artifact proximity detector/tip]
WARNING: no messages received and simulated time is active.
Is /clock being published?

peci1, 2024-12-11:
Please, share more details about what and how you are trying to do.

Your OS, how you launch subt, how you test the pointcloud topic. Also, post the output of the following script from the same console where you test the topic subscription:

```bash
rosparam get /use_sim_time
echo $IGN_TRANSPORT_TOPIC_STATISTICS
echo $IGN_PARTITION
echo $ROS_MASTER_URI
```

jiaminguo1995, 2024-12-16:
> Please, share more details about what and how you are trying to do.
> 
> Your OS, how you launch subt, how you test the pointcloud topic. Also, post the output of the following script from the same console where you test the topic subscription:
> 
> ```shell
> rosparam get /use_sim_time
> echo $IGN_TRANSPORT_TOPIC_STATISTICS
> echo $IGN_PARTITION
> echo $ROS_MASTER_URI
> ```

Thank you for your help.  I am so sorry that I reply late, because I have many examinations these days.  I want to test the algorithm  fast_lio in the underground environment. The first step is to subscribe the point cloud and the IMU information of subt. And then I found that the point cloud was not be published. I tried 2 robots and they have the seem problem. So I guess it may caused by the sensors. The order I tried were as follows:

1.  LC_ALL=C ign launch -v 4 ~/subt_ws/src/subt/submitted_models/explorer_spot_sensor_config_1/launch/example.ign robotName:=X1 ros:=true champ:=true

2. ign launch -v 4 competition.ign worldName:=tunnel_circuit_practice_01 circuit:=tunnel robotName1:=X1 robotConfig1:=X1_SENSOR_CONFIG_1 levels:=false

——————————————————————————————————————————————————————

The outputs of the orders are as follows:

jamie@jamie:~$ rosparam get /use_sim_time
true
jamie@jamie:~$ echo $IGN_TRANSPORT_TOPIC_STATISTICS

jamie@jamie:~$ echo $IGN_PARTITION

jamie@jamie:~$ echo $ROS_MASTER_URI
http://localhost:11311

order 2 and order 3 return nothing.

jiaminguo1995, 2025-03-03:
@peci1 Hello, and sorry to bother you!  I followed your suggestions and input the instructions, and I shared the related output about two months ago. Would you have a chance to take a look when you’re free? Let me know if you need anything else from my side! Thanks!